jamesbowman/i2cdriver #2: Will not compile in MacOS 10.14.3

chappy1978, 2019-02-26:
Hi,
 When I run make to compile the i2cdriver I get these errors. It looks like it's a pointer error.

mkdir -p build/
cc -o build/i2ccl  -I common linux/i2c.c common/i2cdriver.c
common/i2cdriver.c:269:36: warning: passing 'uint8_t [1]' to parameter of type
      'char *' converts between pointers to integer types with different sign
      [-Wpointer-sign]
  if (readFromSerialPort(sd->port, a, 1) != 1)
                                   ^
common/i2cdriver.c:154:41: note: passing argument to parameter 'b' here
size_t readFromSerialPort(int fd, char *b, size_t s)
                                        ^
common/i2cdriver.c:304:38: warning: passing 'uint8_t *' (aka 'unsigned char *')
      to parameter of type 'char *' converts between pointers to integer types
      with different sign [-Wpointer-sign]
  (void)readFromSerialPort(sd->port, devices + 8, 112);
                                     ^~~~~~~~~~~
common/i2cdriver.c:154:41: note: passing argument to parameter 'b' here
size_t readFromSerialPort(int fd, char *b, size_t s)
                                        ^
common/i2cdriver.c:310:31: warning: passing 'uint8_t [2]' to parameter of type
      'const char *' converts between pointers to integer types with different
      sign [-Wpointer-sign]
  writeToSerialPort(sd->port, start, sizeof(start));
                              ^~~~~
common/i2cdriver.c:173:44: note: passing argument to parameter 'b' here
void writeToSerialPort(int fd, const char *b, size_t s)
                                           ^
common/i2cdriver.c:328:33: warning: passing 'uint8_t [65]' to parameter of type
      'const char *' converts between pointers to integer types with different
      sign [-Wpointer-sign]
    writeToSerialPort(sd->port, cmd, 1 + len);
                                ^~~
common/i2cdriver.c:173:44: note: passing argument to parameter 'b' here
void writeToSerialPort(int fd, const char *b, size_t s)
                                           ^
common/i2cdriver.c:331:18: warning: passing 'const uint8_t *' (aka
      'const unsigned char *') to parameter of type 'const char *' converts
      between pointers to integer types with different sign [-Wpointer-sign]
  crc_update(sd, bytes, nn);
                 ^~~~~
common/i2cdriver.c:223:51: note: passing argument to parameter 'data' here
static void crc_update(I2CDriver *sd, const char *data, size_t data_len)
                                                  ^
common/i2cdriver.c:332:1: warning: control reaches end of non-void function
      [-Wreturn-type]
}
^
common/i2cdriver.c:341:33: warning: passing 'uint8_t [1]' to parameter of type
      'const char *' converts between pointers to integer types with different
      sign [-Wpointer-sign]
    writeToSerialPort(sd->port, cmd, 1);
                                ^~~
common/i2cdriver.c:173:44: note: passing argument to parameter 'b' here
void writeToSerialPort(int fd, const char *b, size_t s)
                                           ^
common/i2cdriver.c:342:34: warning: passing 'uint8_t *' (aka 'unsigned char *')
      to parameter of type 'char *' converts between pointers to integer types
      with different sign [-Wpointer-sign]
    readFromSerialPort(sd->port, bytes + i, len);
                                 ^~~~~~~~~
common/i2cdriver.c:154:41: note: passing argument to parameter 'b' here
size_t readFromSerialPort(int fd, char *b, size_t s)
                                        ^
common/i2cdriver.c:343:20: warning: passing 'uint8_t *' (aka 'unsigned char *')
      to parameter of type 'const char *' converts between pointers to integer
      types with different sign [-Wpointer-sign]
    crc_update(sd, bytes + i, len);
                   ^~~~~~~~~
common/i2cdriver.c:223:51: note: passing argument to parameter 'data' here
static void crc_update(I2CDriver *sd, const char *data, size_t data_len)
                                                  ^
common/i2cdriver.c:415:23: warning: passing 'char [8192]' to parameter of type
      'const uint8_t *' (aka 'const unsigned char *') converts between pointers
      to integer types with different sign [-Wpointer-sign]
        i2c_write(sd, bytes, nn);
                      ^~~~~
common/i2cdriver.c:319:44: note: passing argument to parameter 'bytes' here
int i2c_write(I2CDriver *sd, const uint8_t bytes[], size_t nn)
                                           ^
common/i2cdriver.c:429:22: warning: passing 'char [8192]' to parameter of type
      'uint8_t *' (aka 'unsigned char *') converts between pointers to integer
      types with different sign [-Wpointer-sign]
        i2c_read(sd, bytes, nn);
                     ^~~~~
common/i2cdriver.c:334:38: note: passing argument to parameter 'bytes' here
void i2c_read(I2CDriver *sd, uint8_t bytes[], size_t nn)
                                     ^
11 warnings generated.

jamesbowman, 2019-02-26:
OK, switched on -Wall for Linux. I think all are fixed in
9ad4dcda3a1dbfe72f6bb9a8518c0db389b38439
please can you confirm? Thanks.

jamesbowman, 2019-03-03:
Confirmed clean on my 10.13.4
Closing

closed 2019-03-03 by jamesbowman