jamesbowman/i2cdriver #3: Seems to hang ?

lakidd, 2019-03-01:
Hi I've just received my i2cdriver.
I plug it in and it lights up with the default screen, but it doesn't 'do' anything. When running i2ccl with the correct device name, it just seems to hang, i don't get the status line like your example.
Any hints on troubleshooting ?
Thanks

jamesbowman, 2019-03-01:
What device name are you using? Something like:

``/dev/cu.usbserial-XXXXXXXX``

?

lakidd, 2019-03-01:
Aha its /dev/tty.usbserial-WEATHER...when i change to the cu.usbserial-WEATHER, it comes good, thanks.
What is the difference between those nodes ? I've never understood why both appear and for all other arduino, esp stuff the tty nodes work fine.
Any thoughts on the python issue ?

jamesbowman, 2019-03-01:
Yes it is subtle, but this explains tty vs cu in more detail.

https://pbxbook.com/other/mac-tty.html

Please can you raise a new issue for the Python GUI? (WX changed the parameters of GridSizer between versions.)

jamesbowman, 2019-03-02:
OK, think this is closed.

closed 2019-03-02 by jamesbowman