jamesbowman/i2cdriver #6: i2cgui is not functional code.

rhyde, 2019-03-02:
I have fixed a couple obvious problems but I quickly realized that this is at best a stub and very little works.  When will a tool that was advertised be available?
Issues: Wrong number of arguments creating the grid layout.
Can't handle non i2c usb-serial devices.

Randomly will not connect claiming bus is busy

jamesbowman, 2019-03-02:
Good points... this in progress.

When it says the bus is busy, what does the ``i2ccl <port> i`` report?

rhyde, 2019-03-02:
The client tool is responding as expected.

jamesbowman, 2019-03-02:
Hmm, and ``i2ccl <port> d`` produces a reasonable port scan?

jamesbowman, 2019-03-04:
Think it is feature complete now. Tested on Mac, seems to be usable.
Raised #7 to cover the work to expose the programmable pull-ups.

jamesbowman, 2019-03-04:
[image]

closed 2019-03-08 by jamesbowman