jamesbowman/i2cdriver #9: I2CDriver build not working

rlramirez77, 2019-03-08:
1. Need to add a note for users to edit the go.bat with their destination path for i2cdriver-installer.exe output, or have the path resolve for the user automatically
2. Need to mention that Nullsoft Scripting, NSIS is a dependency, user to download and install it
3. Need to reconcile line #63 in i2cdriver.nsi - file "i2ccl.exe" - commenting the line out allows i2cdriver-installer to build, but execution of I2CDriver GUI fails with "Failed to execute script"

jamesbowman, 2019-03-10:
@rlramirez77 when it says "Failed to execute script" what other error text is shown.

rlramirez77, 2019-03-10:
> @rlramirez77 when it says "Failed to execute script" what other error text is shown.

No other message/text, the error is presented in a modal dialog - so it's a typical helpful Windows error message - kek

jamesbowman, 2019-03-10:
OK, to see more error information, do:

    pyinstaller --onefile --icon=../../images/i2cdriver.ico i2cgui.py

(i.e. no ``--windowed`` option). This will leave the console window open and print any error message when the ``.exe`` runs.

rlramirez77, 2019-03-10:
Corresponding output from suggested run (Anaconda is up to date):

(base) C:\Users\Administrator\Downloads\i2cdriver-master\i2cdriver-master\python\samples>pyinstaller --onefile --icon=../../images/i2cdriver.ico i2cgui.py
46 INFO: PyInstaller: 3.4
46 INFO: Python: 3.7.1
46 INFO: Platform: Windows-10-10.0.14393-SP0
46 INFO: wrote C:\Users\Administrator\Downloads\i2cdriver-master\i2cdriver-master\python\samples\i2cgui.spec
46 INFO: UPX is not available.
46 INFO: Extending PYTHONPATH with paths
['C:\\Users\\Administrator\\Downloads\\i2cdriver-master\\i2cdriver-master\\python\\samples',
 'C:\\Users\\Administrator\\Downloads\\i2cdriver-master\\i2cdriver-master\\python\\samples']
62 INFO: checking Analysis
93 INFO: checking PYZ
109 INFO: checking PKG
125 INFO: Bootloader c:\programdata\anaconda3\lib\site-packages\PyInstaller\bootloader\Windows-64bit\run.exe
125 INFO: checking EXE
140 INFO: Building because console changed
140 INFO: Building EXE from EXE-00.toc
249 INFO: SRCPATH [('..\\..\\images\\i2cdriver.ico', None)]
249 INFO: Updating icons from ['..\\..\\images\\i2cdriver.ico'] to C:\Users\ADMINI~1\AppData\Local\Temp\1\tmphuc9ffz2
265 INFO: Writing RT_GROUP_ICON 0 resource with 20 bytes
265 INFO: Writing RT_ICON 1 resource with 75484 bytes
328 INFO: Appending archive to EXE C:\Users\Administrator\Downloads\i2cdriver-master\i2cdriver-master\python\samples\dist\i2cgui.exe
437 INFO: Building EXE from EXE-00.toc completed successfully.

jamesbowman, 2019-03-10:
Looks good... and what does it say when you run ``dist\i2cgui.exe``?

rlramirez77, 2019-03-10:
How do I resolve this dependency?

Traceback (most recent call last):
  File "i2cgui.py", line 11, in <module>
ModuleNotFoundError: No module named 'wx'
[3264] Failed to execute script i2cgui

jamesbowman, 2019-03-10:
It needs wxPython, installation instructions are here

https://www.wxpython.org/pages/downloads/

if you are on Anaconda then it is here:

https://anaconda.org/anaconda/wxpython

rlramirez77, 2019-03-10:
Installed via Anaconda - same line 11 error.

jamesbowman, 2019-03-10:
OK, if you run it from the command-line with:

    python i2cgui.py

then do you see the same message about ``No module named 'wx'``?

rlramirez77, 2019-03-10:
Getting closer to the error:

(base) C:\Users\Administrator\Downloads\i2cdriver-master\i2cdriver-master\python\samples>python i2cgui.py
Traceback (most recent call last):
  File "i2cgui.py", line 14, in <module>
    from i2cdriver import I2CDriver
  File "C:\ProgramData\Anaconda3\lib\site-packages\i2cdriver.py", line 13, in <module>
    import bargraph
  File "C:\Users\Administrator\Downloads\i2cdriver-master\i2cdriver-master\python\samples\bargraph.py", line 10, in <module>
    from i2cdriver import I2CDriver
ImportError: cannot import name 'I2CDriver' from 'i2cdriver' (C:\ProgramData\Anaconda3\lib\site-packages\i2cdriver.py)

jamesbowman, 2019-03-10:
Aha. please update i2cdriver:

    pip install -U i2cdriver

rlramirez77, 2019-03-10:
Better, but 2 things. How do I specify COM9 as an argument to the executable? Like SPIDriver does, I would not throw an error if the i2cdriver hardware is unplugged when the GUI is run. 

(base) C:\Users\Administrator\Downloads\i2cdriver-master\i2cdriver-master\python\samples\dist>i2cgui.exe
Traceback (most recent call last):
  File "i2cgui.py", line 334, in <module>
  File "i2cgui.py", line 209, in __init__
  File "i2cgui.py", line 266, in connect
  File "site-packages\i2cdriver.py", line 47, in __init__
  File "site-packages\serial\serialwin32.py", line 31, in __init__
  File "site-packages\serial\serialutil.py", line 240, in __init__
  File "site-packages\serial\serialwin32.py", line 62, in open
serial.serialutil.SerialException: could not open port 'COM1': PermissionError(13, 'Access is denied.', None, 5)
[7200] Failed to execute script i2cgui

bruceephx, 2019-03-17:
I just received my I2CDriver kit and following the getting-started page I downloaded the Windows Installer program and ran it as Administrator.  It seemed to install the i2cgui.exe and the i2ccl.exe without issue.  I ran the i2cgui.exe as Adminsitrator and I keep getting the "Failed to execute script i2cgui" response.  I'm running Windows 10 Pro x64 latest release.

I just tried the command line "i2ccl.exe" COM2 i" and that gave the response from the I2CDriver board with stats.  So the board appears to be running and communicating.

Ideas or thoughts?

jamesbowman, 2019-03-17:
Hi @bruceephx - What is the output of the "mode" command? Are there other COM ports? That might be a clue as to why i2cgui is failing.

jamesbowman, 2019-03-18:
@rlramirez77 and @bruceephx  I have built the new version of i2cgui at:

https://excamera.com/files/i2cgui.exe

This version is a bit more robust about handling ports, accepts a port on the command-line, and displays a dialog if it catches a fault. Please let me know how it works. Thanks.

rlramirez77, 2019-03-24:
That has it working now, thanks.

jamesbowman, 2019-03-26:
Great, thanks. Closing.

closed 2019-03-26 by jamesbowman