jamesbowman/i2cdriver #11: Unable to see Stop button in 1366 x 768 screen resolution

gojimmypi, 2019-03-10:
perhaps the radio buttons can be vertically pushed closer together a bit, or add a scrollbar for those that have lower resolution screens? I would have submitted PR for this, but could not find the GUI version source code.

jamesbowman, 2019-03-10:
Aha, this is on Windows.
Sure, there are a couple of changes I can make.
btw the source is at ``python/samples/i2cgui.py``

jamesbowman, 2019-03-10:
Yes, pushing the radio buttons closer together looks good. Now fits in 768:

[image]

gojimmypi, 2019-03-10:
yes, that works perfectly! (and looks better, too -  in addition to being able to see the Stop button) 

thank you for the prompt response & I didn't realize Windows GUI was python based.

closed 2019-03-10 by gojimmypi