jamesbowman/i2cdriver #12: Windows app buttons disabled at startup

gojimmypi, 2019-03-25:
I'm test driving an INA219 I2C current sensor. The I2CDriver board sees the device at address x40, but the `read` and `write` buttons are initially disabled.

[image]

If I plug in a second device, Ic2 address = x60, select that one and then go back and select the 0x40 device, the buttons are then enabled.

jamesbowman, 2019-03-26:
This seems to be a difference in radio button behavior between our versions of Windows. Surprising!

If you start with 0x40 attached, then click on the 0x40 radio button, do "read" and "write" light up?

When I start it with a single device (74 in my case) the device map doesn't light up, I have to click it. Then the RADIOBUTTON event happens and the GUI enables the buttons:

[image]

I'm running Windows 10.

gojimmypi, 2019-03-28:
Well, I found a sequence of events to reproduce this. The last time it seemed that the problem continually occurred. 

With the I2CDriver powered off, plug in a device. (I plugged mine into the connector closest to the top of the display).  Launch the software. No items are selected and the buttons are grayed out. Click the ID of your selected item (in my case 40). The `Read` and `Write` buttons become active.

While leaving the I2CDriver board powered up, unplug the single I2C test device. The `Read` and `Write` buttons become inactive and grayed out. The selected device radio button remains selected, but also grayed out.

Re-plug the I2C device. The software recognizes that a device has been plugged in, the radio button selection is no longer grayed out - however clicking on it does not enable the buttons, perhaps because the radio button was already selected? Having only one device - there's no way to un-select and re-select it.

It seems that perhaps the  `Read` and `Write`  buttons are enabled only upon a state change from the radio button not selected to selected.

I'm also running Windows:
[image]

jamesbowman, 2019-03-29:
Aha, that's very clear, thanks.

I committed a fix, and built a test binary here:

https://excamera.com/files/i2cgui.exe

jamesbowman, 2020-01-08:
Fixed, closing.

closed 2020-01-08 by jamesbowman