jamesbowman/i2cdriver #13: Capture Mode

brotherdust, 2019-04-20:
Hi!
THANK YOU for making this device! It's so useful!

While looking through the [documentation](https://i2cdriver.com/), I noticed there's a stub header for "Capture Mode", but nothing below it. Are there any plans for a capture mode to be supported? If so, when? I'd like to contribute to your project where I can, so I'd be happy to write up the documentation for this feature once it's available.

Thanks!

simon-rob, 2019-04-29:
+1
I twas hoping capture mode would allow capturing of I2C frames as monitor mode seems only to so the current frame on the display which is limited in use when debugging. (due to the small screen)

jamesbowman, 2019-04-29:
Hi all, yes, really sorry this isn't. It's coming today.

simon-rob, 2019-04-29:
Great, looking forward to giving it ago.

jamesbowman, 2019-04-30:
Capture to file is implemented in the new release 0.0.5.

https://i2cdriver.com/windows

It's documented on p.18 of the user guide:

https://i2cdriver.com/i2cdriver.pdf

Note that there's no live graphical display at the yet, but the command-line version does print all the traffic as it arrives.

simon-rob, 2019-04-30:
James,

I am using Python 2.7  and I get the follow error when trying to capture:
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "i2cgui.py", line 48, in capture_thr
    c = sd.capture_start(True)
TypeError: capture_start() takes exactly 1 argument (2 given)

If I remove from capture_start() I then get:
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "i2cgui.py", line 52, in capture_thr
    for token in c():
  File "build/bdist.linux-x86_64/egg/i2cdriver.py", line 261, in parser
    for n in nstream():
  File "build/bdist.linux-x86_64/egg/i2cdriver.py", line 258, in nstream
    yield (b >> 4) & 0xf
TypeError: unsupported operand type(s) for >>: 'str' and 'int'

I see that i2cdriver.py has support for PYTHON2, but i guess some of the new capture code needs adapting! (I did have a go, but it kept giving the same error on 258 to my surprise)

Simon

jamesbowman, 2019-05-02:
@simon-rob I think you might be running the latest gui with an old i2cdriver. Did you do

  pip install -U i2cdriver

(or similar) before running the GUI?

simon-rob, 2019-05-02:
James,

I have run python2 setup.py install
I also tried pip install -U i2driver

but I still get:

  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "i2cgui.py", line 47, in capture_thr
    c = sd.capture_start(True)
TypeError: capture_start() takes exactly 1 argument (2 given)

jamesbowman, 2019-05-02:
That definitely looks like a version mismatch because capture_start's signature changed in 0.0.5.

If I do:

```
$ python2
Python 2.7.6 (default, Oct 26 2016, 20:30:19) 
[GCC 4.8.4] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import i2cdriver
>>> i2cdriver.__version__
'0.0.5'
```

do you get the same version 0.0.5?

simon-rob, 2019-05-02:
James,

No, I get version 0.0.4.

I have uninstalled the i2cdriver using  pip uninstall i2driver and then reinstalled it again..

Now no error and the correct version 0.0.5.

Thanks for you help, 

Simon