jamesbowman/i2cdriver #15: Cant start i2cgui.py

logicethos, 2019-05-03:
[image]

jamesbowman, 2019-05-03:
I think that is an old copy of ``i2cgui.py``. If you remove it and re-run ``i2cgui.py`` it should find the right one that pip installed as part of 0.0.5.

logicethos, 2019-05-03:
It was a new install.  I tried re-installing, but it's the same.

`rm /usr/local/bin/i2cgui.py /usr/local/bin/i2cgui.pyc`

```
Uninstalling i2cdriver-0.0.5:
  Would remove:
    /usr/local/lib/python2.7/dist-packages/EDS.py
    /usr/local/lib/python2.7/dist-packages/i2cdriver-0.0.5.dist-info/*
    /usr/local/lib/python2.7/dist-packages/i2cdriver.py
Proceed (y/n)? y
  Successfully uninstalled i2cdriver-0.0.5
```

```
root@stuart:/home/stuart/Projects# pip install i2cdriver
Collecting i2cdriver
Requirement already satisfied: pyserial in /usr/local/lib/python2.7/dist-packages (from i2cdriver) (3.4)
Installing collected packages: i2cdriver
Successfully installed i2cdriver-0.0.5
```

```
root@stuart:/home/stuart/Projects# python --version
Python 2.7.16

```

jamesbowman, 2019-05-03:
OK, thanks. What version of wxPython do you have?

```
$ python
>>> import wx
>>> wx.version()
```

logicethos, 2019-05-03:
`3.0.2.0 gtk3 (classic)`

```
Distributor ID: Ubuntu
Description:    Ubuntu 19.04
Release:        19.04
Codename:       disco

```

jamesbowman, 2019-05-03:
OK, that's the cause.

The app is all written for wx 4.0 - can I ask how you got wx 3.0?

Details are at https://www.wxpython.org/pages/downloads/ - maybe:

    pip install -U wxPython

logicethos, 2019-05-03:
No idea how I got wx 3.0.  I don't use python much.  I tried upgrading but there are further dependency problems. I'll have a go writing a docker container for it.  Thanks anyway.

closed 2019-05-03 by logicethos