jamesbowman/i2cdriver #18: Can't connect in Linux/Ubuntu

kylemallory, 2019-07-07:
I recently purchased this, and was really excited about using it for some of my embedded development. It took a bit to get wxPython installed property on Ubuntu (needed a bunch of additional packages that were in your instructions), but eventually I got it sorted.

Now that it's installed, I can run i2cgui.py, but I can't select a serial port from the drop down. I can see the /dev/ttyUSB0 being created when I connect it, and my user has 'dialout' permissions, but the GUI is almost entirely grayed out, and there is no output on the console.

I'd love to see this work. Any ideas for what might be causing the problems, or steps for further troubleshooting?

Thanks!

[image]

jamesbowman, 2019-07-07:
First question, when you attach the I2CDriver, does the I2CDriver's screen light up, and does it appear as a new device, like ``/dev/ttyUSB0``?

kylemallory, 2019-07-07:
Yes, it powers on, the display shows, and the OS detects the USB device, and registers as /dev/ttyUSB0. I have added my user to the "dialout" group, so permissions appear not to be a problem either.  Is there a 'debug' or 'logging' argument for the i2cgui.py that I can try?

jamesbowman, 2019-07-07:
OK, that narrows it down.

- [ ] Can I confirm that you logged out and back in, so the dialout group change took effect?
- [ ] The command-line tool might show the problem, if you build it and do ``i2ccl /dev/ttyUSB0 i`` what does it show?

kylemallory, 2019-07-07:
I also just noticed; when I start `i2cgui.py` the LCD backlight will brighten for about 2 seconds, and then dim back to normal.  If I close the app, and reopen, the same "bright/dim" change repeats each time I restart it.

kylemallory, 2019-07-07:
I can completely rebooted after adding myself to the group.  I'll try the `i2ccl` tool...

kylemallory, 2019-07-07:
Cloned, built, and ran... 
```
$ ./i2ccl /dev/ttyUSB0 i
uptime 19945  5.388 V  0 mA  23.0 C SDA=0 SCL=1 speed=100 kHz
```
However:
```
$ ./i2ccl /dev/ttyUSB0 d
```
seems to hang indefinitely (I can abort/Ctrl+C).  Calling `i2ccl /dev/ttyUSB0 i` a second time also hung.  I had a i2c LCD display connected.  I remove that, and powered off the i2cdriver, and repowered.  At this point, both queries returned successfully.

I tried a different OLED display, and the device is working correctly, including in i2cgui.py.

There must have been something about the other LCD display that was messing up i2cdriver.  I think we can close this ticket, unless you'd like to do any additional troubleshooting to try and figure out what it going on.

closed 2020-01-08 by jamesbowman