jamesbowman/i2cdriver #19: Issue with GUI on MacOS 10.14.6

joshrawls, 2019-07-31:
I have the device working, occasionally. When it doesn't work I get a few errors in terminal.

I'm still learning programming in general and new to Python, so any help would be appreciated. Thanks!

The first error only appears once:

`Traceback (most recent call last):
  File "/usr/local/bin/i2cgui.py", line 342, in refresh
    devs = self.sd.scan(True)
  File "/Library/Python/2.7/site-packages/i2cdriver.py", line 180, in scan
    d = struct.unpack("112c", self.ser.read(112))
struct.error: unpack requires a string argument of length 112
`
The second error loops until I terminate it:
`Traceback (most recent call last):
  File "/usr/local/bin/i2cgui.py", line 323, in refresh
    self.sd.getstatus()
  File "/Library/Python/2.7/site-packages/i2cdriver.py", line 449, in getstatus
    ccitt_crc) = body.split()
ValueError: need more than 0 values to unpack`

g5pw, 2021-04-10:
Uh, I'm seeing the same error on macOS Big Sur 11.2.3. Did you manage to solve it in any way?

jamesbowman, 2021-04-10:
@g5pw does this error happen intermittently, or every time?

g5pw, 2021-04-11:
Looks like it happens every time, and the GUI locks up :/

joshrawls, 2021-04-23:
I never found a fix, so I stopped using the product and got a good oscilloscope that has decoding.

closed 2021-04-23 by jamesbowman

g5pw, 2021-04-26:
Hey, I think the issue still exists, regardless of @joshrawls' usage of the product. I got one, and it looks nice, and I'd like to use it if possible 😅 I'll try to test the UI part, but from a quick glance it looks like a firmware hang, i.e. the device is not sending any more data.