jamesbowman/i2cdriver #33: Inaccurate data read in capture mode

codehero, 2020-01-21:
I am reading the following data between a Pi and a microcontroller at address 0x31
TX -> 01
RX <- 0d 76 09 01 00 01 01 01 02 00 32 00 00

I shown capture output for 2 identical runs.
The Pi is correctly reading data over i2c.
The i2cdriver LCD display also shows correct data.
The capture mode shows correct data for the first run but not the second.

<START 0x31 WRITE ACK>
<WRITE 0x01 ACK>
<START 0x31 READ ACK>
<READ 0x0d ACK>
<READ 0x76 ACK>
<READ 0x09 ACK>
<READ 0x01 ACK>
<READ 0x00 ACK>
<READ 0x01 ACK>
<READ 0x01 ACK>
<READ 0x01 ACK>
<READ 0x02 ACK>
<READ 0x00 ACK>
<READ 0x32 ACK>
<READ 0x00 ACK>
<READ 0x00 ACK>
<READ 0x10 NACK>
<STOP>

<START 0x31 WRITE ACK>
<WRITE 0x01 ACK>
<START 0x31 READ ACK>
<READ 0x0d ACK>
<START 0x01 WRITE ACK>
<WRITE 0x80 ACK>
<WRITE 0x80 ACK>
<WRITE 0x00 ACK>
<WRITE 0x80 ACK>
<WRITE 0x80 ACK>
<WRITE 0x80 NACK>
<WRITE 0x00 ACK>
<WRITE 0x0c NACK>
<WRITE 0x00 ACK>
<WRITE 0x00 ACK>
<WRITE 0x04 ACK>
<STOP>

[image]

jamesbowman, 2020-01-21:
Is the PI driving I2C at 100 or 400 KHz? If it is 400, might be worth trying at 100.

codehero, 2020-01-21:
The SCL runs at max 25 KHz.