jamesbowman/i2cdriver #36: When connected, i2cbusses merge

jessica-mulein, 2020-02-23:
I have a raspberry pi and had attempted to inspect bus i2c-1 and i2c-6. When connected, device 0x22 on i2c-6 joined bus 1 and vice-versa, according to i2cdetect on my pi-host. Didn't seem to matter whether powered on or not.

Normal:
➜  ~ /usr/sbin/i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- 1c -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- 6a -- -- -- -- --
70: -- -- -- -- -- -- -- 77
➜  ~ /usr/sbin/i2cdetect -y 6
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- 22 -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

When connected:
➜  ~ /usr/sbin/i2cdetect -y 6
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- 1c -- -- --
20: -- -- 22 -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- 6a -- -- -- -- --
70: -- -- -- -- -- -- -- 77
➜  ~ /usr/sbin/i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- 1c -- -- --
20: -- -- 22 -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- 6a -- -- -- -- --
70: -- -- -- -- -- -- -- 77

jamesbowman, 2020-02-23:
Please can you show how the I2CDriver is connected?

One thing to mention is that the 3 ports on the I2CDriver are all physically wired together - they're not separate I2C busses. So when multiple things are connected to I2CDriver, they are all connected together.

jessica-mulein, 2020-02-24:
No, I suppose it makes sense. I looked at it and expected to be able to read 4 separate busses at a time, but those connectors are really more about the output or perhaps separate circuits... Do you suppose you will have a version in the future that doesn't jumper those together so I could record both streams without two devices?

closed 2020-02-24 by jessica-mulein

jamesbowman, 2020-02-24:
Makes sense.

For multiple buses the best setup is multiple I2CDrivers -- they are designed to all work well on the same host. I2C Minis are functionally identical (capture mode is supported), and are available in multiples.

https://i2cdriver.com/mini.html