jamesbowman/i2cdriver #37: SCL low period timing too small?

Sabine-Schnabeltier, 2020-02-24:
According to table 10 of the [I2C specification](https://www.nxp.com/docs/en/user-guide/UM10204.pdf), for 100 kHz operation, SCL has a minimum low period of 4.7 us and a minimum high period of 4.0 us.

Checking the signals generated by the I²CDriver board, SCL is low for roughly 3.5 us and high for the remaining time of the 10 us clock cycle.

The I2C devices I am trying to control specify a 4.7 us clock low time and I suspect that this timing issue may be the reason I am unable to control them.

jamesbowman, 2020-02-24:
Yes, from the I2C is a hardware block in the EFM8BB1 microntroller:

https://www.silabs.com/documents/public/reference-manuals/efm8bb1-rm.pdf

[image]

And there they confirm it. "SCL ... THIGH is typically twice as large as TLOW." Which is very close to what you measured.

I will ask about this on the Silicon Labs support forum. It seems quite clear that if configured for 100 and 400 KHz I2C would be out of spec. This is surprising for an automotive-rated part.

Unfortunately I don't really think there is much that can be done on this. Are you able to say what the I2C device is?

Sabine-Schnabeltier, 2020-02-24:
It is the Cirrus Logic [CS2300-CP](https://4donline.ihs.com/images/VipMasterIC/IC/CIRR/CIRR-S-A0001393877/CIRR-S-A0001393877-1.pdf?hkey=52A5661711E402568146F3353EA87419).

jamesbowman, 2020-02-24:
OK, thanks. I have ordered one from Digikey. I should be able to confirm it in 3-4 days.