jamesbowman/i2cdriver #40: MacOS- TCSANOW, Serial settings: Invalid argument

jessica-mulein, 2020-03-06:
//            i2c_connect(&i2cDriver, "/dev/tty.usbserial-DO029HL5");
            i2c_connect(&i2cDriver, "/dev/cu.usbserial-DO029HL5");

bombs out  at the tcsetatttr call on my mac os Catalina machine
  cfmakeraw(&Settings);
  Settings.c_cc[VMIN] = 1;
  if (tcsetattr(fd, TCSANOW, &Settings) != 0) {
      perror("Serial settings");
    return -1;
  }

Right now my code runs and I can i2c_scan just fine, but my i2c_write's and read's are not working

jessica-mulein, 2020-03-06:
I've actually also discovered that at least some of my problem seems to be with the 3 SparkFun Qwiic LED Sticks plugged into an i2cmini

// i2cdriver has a few things
 ✘    master ●  ./i2ccl /dev/cu.usbserial-DO029HL5 d

--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  --
--  --  --  --  --  --  26  --
--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  --
--  --  42  --  --  --  --  --
--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  --
--  --  --  5b  --  --  --  --
--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  77

// i2cmini connected to LED sticks
 ✘    master ●  ./i2ccl /dev/cu.usbserial-DO029ICO d

never returns

The i2c-mini is connected directly to a string of 3 qwiic LED sticks with nothing else on the line.  Colors are matched up.

jessica-mulein, 2020-03-07:
I'm actually having a really hard time just with i2ccl being unreliable on a line connected directly to three SparkFun Qwiic LED stix using only the short leads provided (less than 12in total wire). I've only gotten this to show up once
```
 ✘   master ●  ./i2ccl /dev/cu.usbserial-DO029ICO d

--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  --
--  --  22  23  24  --  --  --
--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  --
```
some things just don't respond most of the time
```
 ✘    master ●  ./i2ccl /dev/cu.usbserial-DO029ICO x
^C
 ✘    master ●  ./i2ccl /dev/cu.usbserial-DO029ICO i
^C
 ✘    master ●  ./i2ccl /dev/cu.usbserial-DO029ICO x
^C
```
after unplugging the USB and trying again (have tried different cables too):
```
 ✘    master ●  ./i2ccl /dev/cu.usbserial-DO029ICO i
uptime 43  5.000 V  0 mA  27.5 C SDA=1 SCL=1 speed=100 kHz
 ✘   master ●  ./i2ccl /dev/cu.usbserial-DO029ICO d
(...never returns again)
```

Any idea what could be going on?

jamesbowman, 2020-03-07:
That's very strange. My immediate reaction to this kind of thing is to get one to add to the test set, but I see theQwiic LED strips is out of stock.

(1) Maybe the LEDs are drawing so much current that they crash the I2CMini? Maybe connect VCC to a seperate  3.3V supply? Try it with an I2CDriver to view the current measurement. At startup, are all-off or all-on?

(2) On its Sparkfun page people are reporting the the stick itself locks up the I2C bus - buggy firmware. Could it be that the address-scan is putting it into a buggy state? I notice that their library only does I2C writes:

https://github.com/sparkfun/SparkFun_Qwiic_LED_Stick_Arduino_Library/blob/master/src/Qwiic_LED_Stick.cpp

Maybe power-up, and do a single write, copying their code for [setLEDColor](https://github.com/sparkfun/SparkFun_Qwiic_LED_Stick_Arduino_Library/blob/master/src/Qwiic_LED_Stick.cpp#L53-L67):

i2ccl /dev/cu.usbserial-DO029ICO w 0x23 0x72,0x00,0xff,0x00 p

meaning COMMAND_WRITE_ALL_LED_COLOR,0,255,0. It should set all LEDs to green.

jessica-mulein, 2020-03-07:
Switching the configuration around, I can enumerate it on the full i2cdriver repeatedly

   master ●  ./i2ccl /dev/cu.usbserial-DO029HL5 d

--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  --
--  --  22  23  24  --  --  --
--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  --

   master ●  ./i2ccl /dev/cu.usbserial-DO029HL5 d

--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  --
--  --  22  23  24  --  --  --
--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  --
--  --  --  --  --  --  --  --
```
and more.. 

Maybe the i2cmini is different?

jamesbowman, 2020-03-07:
I2CDriver and I2CMini are build from the same code base. What does 'i' report as the current usage?

jessica-mulein, 2020-03-07:
Yeah that's about what I'm seeing. Sweet.. so.. it's not your problem.

```
   master ●  ./i2ccl /dev/cu.usbserial-DO029ICO i
uptime 2294  5.000 V  0 mA  35.2 C SDA=1 SCL=1 speed=100 kHz
   master ●  ./i2ccl /dev/cu.usbserial-DO029HL5 i
uptime 7841  5.240 V  10 mA  28.7 C SDA=1 SCL=1 speed=100 kHz
```
LED sticks on the latter.

jessica-mulein, 2020-03-07:
None of these were successful in lighting up any LEDs off the i2cdriver full :/ even after a full power on reset. *shrug* OK well- moving on without the LED's in my project I think then!
```
   master ●  ./i2ccl /dev/cu.usbserial-DO029HL5 w 0x23 0x72,0x00,0xff,0x00 p
   master ●  ./i2ccl /dev/cu.usbserial-DO029HL5 w 0x23 0x72,0x00,0xff,0x00 p
   master ●  ./i2ccl /dev/cu.usbserial-DO029HL5 w 0x23 0x72,0x00,0xff,0x00 p
   master ●  ./i2ccl /dev/cu.usbserial-DO029HL5 w 0x22 0x72,0x00,0xff,0x00 p
   master ●  ./i2ccl /dev/cu.usbserial-DO029HL5 w 0x24 0x72,0x00,0xff,0x00 p
   master ●  ./i2ccl /dev/cu.usbserial-DO029HL5 w 0x24 0x72,0x00,0xff,0x00 p
   master ●  ./i2ccl /dev/cu.usbserial-DO029HL5 w 0x24 0x72,0x00,0xff,0x00 p
```

jessica-mulein, 2020-03-07:
If you have an email address you don't mind me sending a photo of the setup to, I can do so (with the output on the screen of the i2cdriver full)[](url)

jamesbowman, 2020-06-01:
Seems like the peripheral is locking up - and Sparkfun retired it so there's no obvious way of debugging this.

OK to close?

jessica-mulein, 2020-06-01:
Yep!

closed 2020-06-01 by jessica-mulein