jamesbowman/i2cdriver #44: IndexError if no IO connected

markddavidoff, 2020-05-18:
>>> i2cdriver.I2CDriver("/dev/tty.usbserial-DO02C6S9").scan()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.7/site-packages/i2cdriver.py", line 109, in __init__
    r = self.__echo(c)
  File "/usr/local/lib/python3.7/site-packages/i2cdriver.py", line 140, in __echo
    return r[0]
IndexError: index out of range

jamesbowman, 2020-05-18:
It looks like the I2CDriver has hung at this point. That's why the initial 'echo' check fails. Pretty fundamental.

Starting from a freshly plugged-in I2CDriver, what gets it into this hung state? Or does it do this immediately after power-on?

markddavidoff, 2020-05-18:
I get this anytime something is not plugged into the i2cdriver, immediatley after power on. if i have something connected i'll get the  i2c grid output.

jamesbowman, 2020-05-18:
Interesting. It might be a hardware problem with the I2CDriver's pullup resistors. If you do (from a fresh power-on with nothing connected):

    i2 = i2cdriver.I2CDriver("/dev/tty.usbserial-DO02C6S9")
    i2.setpullups(0x3f)
    i2.scan()

do you see the same issue?

markddavidoff, 2020-05-21:
thanks for the quick response and sorry for the delay, here's the output i get:
yeah seems to be outputting :
```
-- -- -- -- -- -- -- --
-- -- -- -- -- -- -- --
-- -- -- -- -- -- -- --
-- -- -- -- -- -- -- --
-- -- -- -- -- -- -- --
-- -- -- -- -- -- -- --
-- -- -- -- -- -- -- --
-- -- -- -- -- -- -- --
-- -- -- -- -- -- -- --
-- -- -- -- -- -- -- --
-- -- -- -- -- -- -- --
-- -- -- -- -- -- -- --
-- -- -- -- -- -- -- --
-- -- -- -- -- -- -- --
[]
```
consistently now, odd, because i was getting that error consistently before as well. thanks!

jamesbowman, 2020-06-01:
OK, thanks. Closing.

closed 2020-06-01 by jamesbowman