jamesbowman/i2cdriver #46: Difficulty using with Mac OS X 10.15.5

joshrawls, 2020-06-01:
I had this working at one time, but it's been awhile.

At first it was only showing up as a "Composite Device" in System Information. I re-installed the FTDI drivers and now it shows up as "FT230X Basic UART" so I think I'm making some progress.

When I try to load the GUI no device shows in the drop-down area to select the I2CDriver com port.

What else am I missing? Keep in mind I'm a Linux/newb so if you have command line instructions for me please let me know what each command is doing so I can learn while I fix.

Thanks!

joshrawls, 2020-06-01:
I have an addition concerning Windows 10 Pro 64-bit.

I can get the device to show up as a "USB Serial Converter" in Device Manager, so I think I have the drivers installed correctly.

I installed the Windows version of I2CDriver and I have the same results as on the Mac. No device shown in the drop-down menu.

What am I doing wrong in both OS's?

jamesbowman, 2020-06-01:
Does the serial number in Device Manager match the one on the bottom of the I2CDriver?

joshrawls, 2020-06-01:
It shows the correct serial number in System Information.

I was able to get it working on a separate Windows 10 laptop. So I know the device is still good, but this isn't a permanent solution. On that laptop it shows up in Device Manager as a COM3 port and does not show the serial number anywhere I could find. I'm also not super familiar with Windows 10 so I may be missing it somewhere.

Update: I just looked for it as a COM port on my other Windows install that wasn't previously working. It was showing as COM5 but wouldn't show up in the GUI. I changed it to COM6 and now it is working under windows. Still no luck on the Mac.