jamesbowman/i2cdriver #47: Trying to run i2cDriver.exe on windows 10

txf-, 2020-06-21:
I installed the application using the installer. Unfortunately I have issues using it.

Either the application window does not appear, but the app remains running in the process list.

Or the application window does show up but I can't actually select any COM ports.
[image]

jamesbowman, 2020-06-22:
OK, there seem to be two issues here.

1. Running the current i2cgui.py Python doesn't connect to COM ports

2. The packaged version of the GUI does not launch reliably

Can we start with (1)... a few quick questions

 * what version of Windows are you running this on? 
 * when you run i2ccl.exe, do you get output for the "i" command, e.g.
       ```i2ccl.exe COM6 i```
 * are there other serial devices attached?

scottbob09, 2020-06-23:
uptime 365  4.856 V  15 mA  30.6 C SDA=0 SCL=1 speed=100 kHz

jamesbowman, 2020-06-24:
OK, thanks.

When you start the GUI on the command-line, you can supply the port, which avoids the GUI's discovery process, e.g.

    python i2cgui.py COM6

This should be worth a try.

txf-, 2020-06-24:
Supplying the com port as an argument works

scottbob09, 2020-06-25:
For me as well

madgrizzle, 2021-01-28:
I have the same issue with the Windows 10 application.  Is there any plans on updating it?  running 

`i2ccl.exe COM6 i` 

responds fine, but the Windows 10 gui application doesn't populate the comports.

jamesbowman, 2021-01-28:
If you start the gui with the port on the command-line

    i2cgui COM6

does it work?

madgrizzle, 2021-01-28:
I will try tonight.  I've only tried using the windows icon.

vonkonow, 2021-02-01:
I’m having the same problem. Port won’t show up in gui. Doesn’t help if I run "i2cgui COM7"
(“i2ccl.exe COM7 I” works: “uptime 6  5.167 V  0 mA  25.9 C SDA=1 SCL=1 speed=100 kHz”)

jamesbowman, 2021-02-01:
@vonkonow 

When you run "i2cgui COM7" from the command-line, what does it do?

vonkonow, 2021-02-02:
> @vonkonow
> 
> When you run "i2cgui COM7" from the command-line, what does it do?

The program started, but port was empty (like the first image in this thread). However after running the program as admin and a reboot later, everything seems to work fine now.

RoryMacKean, 2021-04-17:
I'm seeing the same issue as @vonkonow and @madgrizzle.

I'm running Windows 10 (in VirtualBox, on a Mac Catalina host).
`i2ccl COM3 i` returns:
`uptime 448  5.152 V  0 mA  28.2 C SDA=0 SCL=0 speed=100 kHz`

but when I run the gui, either by double-clicking or running `i2cgui COM3` I get an image the same as the one at the top.  This also happens if I try those things as Administrator.

Unfortunately for me, though, a reboot hasn't fixed the issue.

MarkLuinenburg, 2021-05-25:
Hey all, 

I stumbled upon the same issue using windows 10. I went into the device manager , set it up as generic COM port. Then when it was listed there i updated the driver (as it still wasnt working ) with the ft230xq ( https://ftdichip.com/products/ft230xq/ ) driver. 

This resolved the issue. So my guess is that windows mislabeled it due to missing drivers. I just forced it to be recognized as an COM port and then allowed the proper driver to interact. 

Hope this helps people with similar problems!