jamesbowman/i2cdriver #48: i2ccl bug - Windows 10 command line

markhen57, 2020-06-30:
This command works once and then i2ccl returns to prompt with no action (for all i2ccl commands, including "d", "i" etc.):

>i2ccl \\.\COM11 w 0x15 0x03 r 0x15 2 p

Only fix I've found is to unplug the i2cdriver hardware and reconnect.

This command results in continued operation of i2ccl and i2cdriver without issue:

>i2ccl \\.\COM11 w 0x15 0x03 r 0x15 2

i.e. no trailing "p" for stop. The user guide examples show a trailing "p", which doesn't work after first time.

Would also request adding Windows port naming info to User Guide as it took me a long time to figure out "\\.\COM11" !!!?!!

jamesbowman, 2020-07-01:
OK, thanks. Yes those extra "p" commands are illegal because the "r" command does an implicit "p".

Ideally the tool/library would prevent this, but for now I've fixed the document.

Also have added description of the Windows COM10 naming to the user guides for SPIDriver and I2CDriver:

[image]

markhen57, 2020-07-01:
Much appreciated! Great tool! Thanks!

closed 2020-08-09 by jamesbowman