jamesbowman/i2cdriver #51: i2cdriver.pdf documentation does not cover error code for "transmit status info" command

lukehutch, 2020-08-05:
The documentation at 

https://i2cdriver.com/i2cdriver.pdf

specifies that the `'?'` command returns a status string of the form

```
[i2cdriver1 DO01JUOO 000000061 4.971 000 23.8 I 1 1 100 24 ffff ]
```

However, if the I2C Mini gets into some sort of error state, it can return simply

```
1
```

I assume this is an error code? This needs to be documented.

lukehutch, 2020-08-05:
It also seems to return all zeros if the device is still being initialized:

```
00000000000000000000000000000000000000000000000000000000000000000000000000000000
```

jamesbowman, 2020-08-05:
Hmm, the two ends (I2CMini and the host) can get out of sync. Then it gets confusing - the PC doesn't get the response it requests, as above.

From a cold start, sending '?' will always result in the 80-byte status block being returned (p.31).

There is a situation where I2CMini returns a block of zeroes, and that is a "scan devices" command 'd' (p.36). Could it be that a 'd' is sent earlier, and the PC did not read all 112-bytes that 'd' sends back?

lukehutch, 2020-08-05:
For the '1'  response, that's probably what happened.

For the response that was all zeroes, that was right after plugging the device in and immediately running my code from scratch, in which the info command is the first thing to run. So I assume that a scan command is implicitly run on boot?

jamesbowman, 2020-08-05:
Hmm, at boot all the I2CMini does is wait for a command over USB.

This is what the i2ccl does at boot:

https://github.com/jamesbowman/i2cdriver/blob/master/c/common/i2cdriver.c#L243-L269

- First it sends 64 noop ('@') commands, to clear out any partial command that the I2CMini might be executing.
- Then it sends a few echo 'e' commands to confirm the serial port is in sync and running.
- Then it does '?' and parses the resulting status string.

lukehutch, 2020-08-06:
I can't replicate the zeroes behavior, so I'll go ahead and close this.

closed 2020-08-06 by lukehutch