jamesbowman/i2cdriver #52: Feature: setting a default startup mode

jamesbowman, 2020-08-09:
Some users want it always to enter monitor mode, and currently have to use a tool to do it.
It would be good to have a "set default mode" operation that sets it into a particular mode at startup.

A booted i2cdriver in monitor/capture mode goes into mode 'D' as soon as the first character is received.

So a connecting PC's first transmitted character would put it back into driver ('D') mode.

rnsi, 2020-08-10:
I would like this feature. Can you just save the current mode to EEPROM and read it at next boot?

rnsi, 2020-08-10:
And could we also save the pull-ups?

AlvaroHermida, 2020-08-30:
Yes, it will be a great improvement for i2cdriver. Without it, when there is no PC available the device becomes just a dead brick in the toolbox...

jamesbowman, 2020-08-30:
@AlvaroHermida ... one question. How would you power it in this scenario?

rnsi, 2020-08-30:
Usb power pack

AlvaroHermida, 2020-08-30:
Yes, these practical power bricks are everywhere nowadays.

fake-name, 2020-09-29:
I don't understand why you'd ever want it to *not* enter monitor mode on power up.

It seems that the device should boot into monitor mode, and only be switched *out* of it by a command. After all, until a host computer talks to it over the serial interface, it's currently useless.

I bought one of these thinking it would operate more like an oscilloscope (e.g. a standalone instrument). If I wanted to have to futz with stuff on my computer to make it work, I'd just use the logic analyzer I already have.

rnsi, 2020-11-22:
Any update on this? I think this is a necessary feature. Thanks.

gardnmc, 2021-01-15:
Hi James, great little device.  Any news on this feature?  To be able to see the results of a bus scan would make this a very useful production pcb testing tool.  Thanks.

hitandrepair, 2021-03-08:
any update on this? would be handy to have for my project. I am using this to check proper wiring for lithium ion batteries for a hospital setting.

rnsi, 2021-03-08:
I'd love to see this.

AlvaroHermida, 2021-03-08:
I'm no longer expecting it...

hitandrepair, 2021-03-08:
> I'd love to see this.

Vendors supplied us batteries with crossed wires. They would work for a day then the medical equipment would shut off and constantly reboot because the clock and data lines were mixed. I initially made the breadboard to fit a certain size but that was wrong. I still have to convert lines around but you get the main idea in how this works. You plug the battery in, and the address of chip blinks. That’s how you know it works. If it doesn’t do anything then the wires are wrong. Pretty cool technology. Vcc from the board is cut off, and without taking the outside shell off, you have no other way of testing this without risking a machine to malfunction. They use this vendor to save about 20k every 2 years. How to test lithium-ion battery clk and sda bus lines with i2c

GOTO-GOSUB, 2021-08-25:
I only ever use mine in monitor mode and would certainly welcome the ability to change its default state.

rnsi, 2021-08-25:
Still holding my breath for this...

rnsi, 2022-01-27:
It's been a while, any chance we can get this feature? I see it's also requested in the comments on the Amazon page. It would be so much more convenient and usable if I didn't have to use a PC just to use monitor mode.

Hoping for an update...

EvanKrall, 2025-08-09:
I'd love to see this -- I also envision myself mostly using the device in monitor mode.

I could take a stab at implementing this myself if there were instructions on how to compile the firmware (#26 / #62).