jamesbowman/i2cdriver #54: Bus arbitration

limi97, 2020-10-14:
Can you tell me about anything I need to do specifically to handle bus arbitration? What is supported? If I do the start command the arbitration bit is always set. How can I change this?

jamesbowman, 2020-10-14:
[image]

This bit should definitely be zero. Being set suggests that either SCL or SDA is tied low. As a first test, I suggest you:

1. remove power from I2CDriver
2. remove all I2C peripherals from I2CDriver
3. apply power to I2CDriver

And check if the **start** still leaves ARB high. If it does, the I2CDriver has a hardware fault, and I can replace it. If not, there is a problem with the I2C peripheral not releasing the bus.

limi97, 2020-10-14:
I tested this. The bit is zero if there is no peripheral connected to the driver. But if I connect a peripheral the bit goes one. I connected several options.

jamesbowman, 2020-10-14:
Aha, thanks. On the software side, are you using the C library, Python, or something else?

jamesbowman, 2020-10-14:
Also (sorry to have to ask this, but it has happened before) have you checked your hookup that SDA and SCL re not swapped? SCL is yellow, SDA is blue.

limi97, 2020-10-14:
Yes I have it the right way. Actually now it works. I did again what you described and now it works perfectly fine.

limi97, 2020-10-14:
Thanks for your help

closed 2020-10-14 by limi97