jamesbowman/i2cdriver #61: Windows 11 not playing nice with I2CDriver application

PDE-Paul, 2022-03-24:
Hi,

I recently upgraded to Windows 11 64 bit and now my I2CDriver application (GUI) does not list out COMMPORTS.

My results are exactly the same as this post https://github.com/jamesbowman/i2cdriver/issues/47 except installing the FTDI drivers does not resolve the issue, command line works fine. IO have tried all the suggestions in issue 47 with the same results.

Any advise would be greatly appreciated.

Kind Regards
Paul Brink

jamesbowman, 2022-03-24:
So to confirm: after the upgrade, ``i2ccl`` still works OK, but the I2CGui does not list out the available ports?

PDE-Paul, 2022-03-24:
Yes, That is correct. Using i2ccl, I am able to enable monitoring mode. I have tried i2cgui COM6 and the GUI is displayed with zero items in the COMM dropdown, exactly the same result if I execute the i2cdriver shortcut. I have tried executing as administrator with the same result.

Yet Windows 10 works, I have tried a second laptop earlier.