jamesbowman/i2cdriver #70: i2cdriver pip package fails to install

zapta, 2025-05-13:
Pip install of ``i2cdriver`` package fails with an error message and exit code. Please see details below.

**ERROR: For req: i2cdriver. Invalid script entry point: <ExportEntry i2cgui = samples.i2cgui:None []> - A callable suffix is required. Cf https://packaging.python.org/specifications/entry-points/#use-for-scripts for more information.**

```
$ python --version
Python 3.13.1

$ pip uninstall i2cdriver
WARNING: Skipping i2cdriver as it is not installed.
/projects$ pip install --no-cache-dir  i2cdriver
Collecting i2cdriver
  Downloading i2cdriver-1.0.5-py3-none-any.whl.metadata (558 bytes)
Requirement already satisfied: pyserial in /Library/Frameworks/Python.framework/Versions/3.13/lib/python3.13/site-packages (from i2cdriver) (3.5)
Downloading i2cdriver-1.0.5-py3-none-any.whl (8.1 kB)
Installing collected packages: i2cdriver
ERROR: For req: i2cdriver. Invalid script entry point: <ExportEntry i2cgui = samples.i2cgui:None []> - A callable suffix is required. Cf https://packaging.python.org/specifications/entry-points/#use-for-scripts for more information.
/projects$ echo $?
1
```

tayken, 2025-05-13:
Just so that everything is together, here is the output from my test:

```
(.venv) tayken@wadew:~/test$ pip install greenpak                                                                                                                                             
Collecting greenpak                                                                            
  Downloading greenpak-0.0.29-py3-none-any.whl.metadata (2.1 kB)             
Collecting hackpyratebus>=0.0.6 (from greenpak)                                                
  Using cached hackPyrateBus-0.0.6-py3-none-any.whl.metadata (4.1 kB)                                                                                                                         
Collecting i2c-adapter>=0.0.4 (from greenpak)                                                                                                                                                 
  Using cached i2c_adapter-0.2.21-py3-none-any.whl.metadata (3.0 kB)                                                                                                                          
Collecting i2cdriver>=1.0.1 (from greenpak)                                                                                                                                                   
  Using cached i2cdriver-1.0.5-py3-none-any.whl.metadata (558 bytes)                                                                                                                          
Collecting intelhex>=2.3.0 (from greenpak)                                                     
  Using cached intelhex-2.3.0-py2.py3-none-any.whl.metadata (2.7 kB)
Collecting smbus2==0.4.3 (from greenpak)                                                       
  Using cached smbus2-0.4.3-py2.py3-none-any.whl.metadata (7.1 kB)
Collecting pyserial (from hackpyratebus>=0.0.6->greenpak)                                                                                                                                     
  Using cached pyserial-3.5-py2.py3-none-any.whl.metadata (1.6 kB)
Downloading greenpak-0.0.29-py3-none-any.whl (23 kB)                                                                                                                                          
Using cached smbus2-0.4.3-py2.py3-none-any.whl (11 kB)                                         
Using cached hackPyrateBus-0.0.6-py3-none-any.whl (43 kB)                                      
Using cached i2c_adapter-0.2.21-py3-none-any.whl (8.4 kB)                                                                                                                                     
Using cached i2cdriver-1.0.5-py3-none-any.whl (8.1 kB)                                         
Using cached intelhex-2.3.0-py2.py3-none-any.whl (50 kB)                  
Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)                                         
Installing collected packages: smbus2, pyserial, intelhex, i2cdriver, i2c-adapter, hackpyratebus, greenpak
ERROR: For req: i2cdriver>=1.0.1. Invalid script entry point: <ExportEntry i2cgui = samples.i2cgui:None []> - A callable suffix is required. Cf https://packaging.python.org/specifications/en
try-points/#use-for-scripts for more information.                                                                                                                                             
(.venv) tayken@wadew:~/test$ pip install greenpak                                              
Collecting greenpak                                                                            
  Using cached greenpak-0.0.29-py3-none-any.whl.metadata (2.1 kB)                              
Collecting hackpyratebus>=0.0.6 (from greenpak)                                                
  Using cached hackPyrateBus-0.0.6-py3-none-any.whl.metadata (4.1 kB)                          
Collecting i2c-adapter>=0.0.4 (from greenpak)                                                  
  Using cached i2c_adapter-0.2.21-py3-none-any.whl.metadata (3.0 kB)                           
Requirement already satisfied: i2cdriver>=1.0.1 in ./.venv/lib/python3.12/site-packages (from greenpak) (1.0.5)
Requirement already satisfied: intelhex>=2.3.0 in ./.venv/lib/python3.12/site-packages (from greenpak) (2.3.0)
Requirement already satisfied: smbus2==0.4.3 in ./.venv/lib/python3.12/site-packages (from greenpak) (0.4.3)
Requirement already satisfied: pyserial in ./.venv/lib/python3.12/site-packages (from hackpyratebus>=0.0.6->greenpak) (3.5)
Using cached greenpak-0.0.29-py3-none-any.whl (23 kB)  
Using cached hackPyrateBus-0.0.6-py3-none-any.whl (43 kB)
Using cached i2c_adapter-0.2.21-py3-none-any.whl (8.4 kB)
Installing collected packages: i2c-adapter, hackpyratebus, greenpak
Successfully installed greenpak-0.0.29 hackpyratebus-0.0.6 i2c-adapter-0.2.21
```

My guess is [this line](https://github.com/jamesbowman/i2cdriver/blob/master/python3/pyproject.toml#L21) is the problem. But that's just a guess.

tayken, 2025-05-21:
Actually I think this issue is already resolved in `pyproject.toml` file with https://github.com/jamesbowman/i2cdriver/commit/3e5eeb5a4f0c0b9e943ab6a1a639e17e647776b5#diff-b3c69db0a2f6847390de922e6a18901222bd4ca7d740522588d8a039b6df6eaa. The only issue is version number stayed the same, so if a new version, aka `1.0.6` is build and pushed, it should solve this issue.

tayken, 2025-05-22:
No need to do anything in `greenpak` as `greenpak` already uses the latest version of `i2cdriver` which is 1.0.5. `i2cdriver` package needs to be bumped up to 1.0.6 and pushed.

zapta, 2025-05-22:
Ok, thanks.

jamesbowman, 2025-06-30:
Believe this is fixed in new release 1.0.6

https://pypi.org/project/i2cdriver/1.0.6/

tayken, 2025-07-01:
Yes, the new release solves this issue. I think we close this.

* Installing it by itself:
```
(.venv) tayken@wadew:~$ pip install i2cdriver
Collecting i2cdriver
  Using cached i2cdriver-1.0.6-py3-none-any.whl.metadata (558 bytes)
Collecting pyserial (from i2cdriver)
  Using cached pyserial-3.5-py2.py3-none-any.whl.metadata (1.6 kB)
Using cached i2cdriver-1.0.6-py3-none-any.whl (8.1 kB)
Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)
Installing collected packages: pyserial, i2cdriver
Successfully installed i2cdriver-1.0.6 pyserial-3.5
```

* Installing it as a dependency:
```
(.venv) tayken@wadew:~$ pip install greenpak
Collecting greenpak
  Using cached greenpak-0.0.30-py3-none-any.whl.metadata (2.1 kB)
Collecting hackpyratebus>=0.0.6 (from greenpak)
  Using cached hackPyrateBus-0.0.6-py3-none-any.whl.metadata (4.1 kB)
Collecting i2c-adapter>=0.0.4 (from greenpak)
  Using cached i2c_adapter-0.2.21-py3-none-any.whl.metadata (3.0 kB)
Collecting i2cdriver>=1.0.1 (from greenpak)
  Downloading i2cdriver-1.0.6-py3-none-any.whl.metadata (558 bytes)
Collecting intelhex>=2.3.0 (from greenpak)
  Using cached intelhex-2.3.0-py2.py3-none-any.whl.metadata (2.7 kB)
Collecting smbus2==0.4.3 (from greenpak)
  Using cached smbus2-0.4.3-py2.py3-none-any.whl.metadata (7.1 kB)
Collecting typing-extensions>=4.7.1 (from greenpak)
  Using cached typing_extensions-4.14.0-py3-none-any.whl.metadata (3.0 kB)
Collecting pyserial (from hackpyratebus>=0.0.6->greenpak)
  Using cached pyserial-3.5-py2.py3-none-any.whl.metadata (1.6 kB)
Using cached greenpak-0.0.30-py3-none-any.whl (23 kB)
Using cached smbus2-0.4.3-py2.py3-none-any.whl (11 kB)
Using cached hackPyrateBus-0.0.6-py3-none-any.whl (43 kB)
Using cached i2c_adapter-0.2.21-py3-none-any.whl (8.4 kB)
Downloading i2cdriver-1.0.6-py3-none-any.whl (8.1 kB)
Using cached intelhex-2.3.0-py2.py3-none-any.whl (50 kB)
Using cached typing_extensions-4.14.0-py3-none-any.whl (43 kB)
Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)
Installing collected packages: smbus2, pyserial, intelhex, typing-extensions, i2cdriver, i2c-adapter, hackpyratebus, greenpak
Successfully installed greenpak-0.0.30 hackpyratebus-0.0.6 i2c-adapter-0.2.21 i2cdriver-1.0.6 intelhex-2.3.0 pyserial-3.5 smbus2-0.4.3 typing-extensions-4.14.0
```