jamesbowman/i2cdriver #72: Register read values of 0x3B and 0x0A print in ASCII

CollinSchairer, 2025-06-30:
Had a confusing afternoon when dumping registers of a device and seeing “;” and “\n” in the console among normal hex codes. Seems these values 0x3B and 0x0A are not being printed/returned as hex code but instead converting to ASCII characters. 

Not sure if there’s anything that can be done here as I suspect this is weird behavior from python’s handling of the Bytes object.

jamesbowman, 2025-06-30:
Just checking, is this the kind of thing that you're seeing.

```
>>> bytes([0xf6, 0x0a])
b'\xf6\n'
```

If so, that's Python's unhelpful way of showing a `bytes` object. `bytes` object has a `hex()` method that I use to get a proper dump:

```
>>> x=bytes([0xf6, 0x0a])
>>> x.hex(' ')
'f6 0a'
```

CollinSchairer, 2025-06-30:
Yep, that’s exactly the behavior I saw. Thanks for clarifying, fortunately the display on the I2CDriver was able to show me the proper hex values on the bus.

closed 2025-06-30 by CollinSchairer